westlicht/performer-hardware #7: Consider replacing DAC8568C with DAC8568A

westlicht, 2017-11-22:
The C grade have 5V full scale output, but due to VCC at 3.3V we only use half range at 2.5V and loose 1 bit of accuracy. We could replace with DAC8568A which has full range at 2.5V, so this would allow us to use all bits, however, we use some accuracy due to lower grade of internal reference.

8bitPig, 2019-08-20:
Aren't we loosing 8bits of accuracy? 
Still worth the higher temp drift even considering the 33k/22k Voltage divider (each resistor has 100-150ppm) for Vbias which then gets gained times 4.

btomwhitehead, 2020-01-29:
Resurrecting an old issue here but thought I its a pertinent place to raise it. If you dont think so, Im happy to delete my comment.

I made the mistake of getting a DAC8568B, which like A, doesn't have the 2x output gain. I am not worried about the initialisation value being at midpoint given the firmware appears to init the Dac to 0v regardless.

So the two options are:
- Replace the DAC (and there will be an effective 15bit resolution)
- Update firmware to reflect the reduced output gain.

[image]

From my understanding of this issue and the effective output in the datasheet, it seems that the leading bit is irrelevant because the VCC is less than the 5v range. My thought it then that the gain can be made up with writing double the DATA value to the DAC. If this is true, if we reduce the right data bitwise shifts in the Dac write by 1, we can effectively make up the range:
```
void Dac::writeDac(uint8_t command, uint8_t address, uint16_t data, uint8_t function) {
    uint8_t b1 = command;
    uint8_t b2 = (address << 4) | (data >> 12);
    uint8_t b3 = data >> 4;
    uint8_t b4 = (data & 0xf) << 4 | function;
  ...
```
to 
```
void Dac::writeDac(uint8_t command, uint8_t address, uint16_t data, uint8_t function) {
    uint8_t b1 = command;
    uint8_t b2 = (address << 4) | (data >> 11);
    uint8_t b3 = data >> 3;
    uint8_t b4 = ((data << 1) & 0xf) << 4 | function;
  ...
```
Do you think this could work? I am going to give it a test in the coming days. Or do you know a better place (this is no doubt a hack) to make such a change?

🙏 🙏 🙏

westlicht, 2020-03-29:
Hey, I only saw this comment now. I think the change you did is valid, shifting the output by one bit. I might consider adding that to the hardware config so people can actually use the cheaper DAC if the wish to. I haven't measured the difference in accuracy, would have to first make a performer with the A grade DAC.

btomwhitehead, 2020-03-29:
Great!

Yeah, I have been using it since without any complaints or noticeable drift. 

Thanks for the response.

Timnuge, 2020-04-10:
Hello! 

My Performer is built with DAC8568A🤔 A way to switch would make my life easy or, could you please share with me update file? Any help would be greatly appreciated!

btomwhitehead, 2020-04-10:
@Timnuge as per above notes, heres the changes I made to `src/platform/stm32/drivers/Dac.cpp`:

```
--- a/src/platform/stm32/drivers/Dac.cpp
+++ b/src/platform/stm32/drivers/Dac.cpp
@@ -72,9 +72,9 @@ void Dac::write() {

 void Dac::writeDac(uint8_t command, uint8_t address, uint16_t data, uint8_t function) {
     uint8_t b1 = command;
-    uint8_t b2 = (address << 4) | (data >> 12);
-    uint8_t b3 = data >> 4;
-    uint8_t b4 = (data & 0xf) << 4 | function;
+    uint8_t b2 = (address << 4) | (data >> 11);
+    uint8_t b3 = data >> 3;
+    uint8_t b4 = ((data << 1) & 0xf) << 4 | function;

     gpio_clear(DAC_PORT, DAC_SYNC);--- a/src/platform/stm32/drivers/Dac.cpp
```

Would have pushed a custom branch to the main repo but (very sensibly) I don't have the requisite permissions.