westlicht/performer-hardware #28: Hardware supervisor

forestcaver, 2019-10-01:
As you are revising the hardware, in lieu of other solutions, it may be worth adding a hardware supervisor eg a mic803 (along with its 10k resistor and 100nf cap).
I added one to my performer and it resolved all my troublesome bootup issues. Not had a single failed boot since adding it.

[image]
[image]

grantHarris, 2020-11-23:
I have to echo this. I built a performer and it doesn't seem to come up properly on about 1 in 10 boots. Blank or white noise on the screen. Unresponsive.

grantHarris, 2020-11-23:
I'd be happy to submit a PR for this if the approach is supported

forestcaver, 2020-11-23:
A lot of people (see muff wiggler thread and fb group) have adopted this approach without any problems. I’ve had no issue with mine since adding it. (I tried fw mods first with no joy). There are board revisions floating around with this mod now. I suspect that, for Simon, revising the hardware is a pretty expensive thankless task as he’d need to order, build and then test any revision; so I totally understand why it hasnt been revised. The best way may be for you to just fork it... there are quite a few hardware mods that would be great to see one day...