westlicht/performer-hardware #29: Rev 1 board dead.

orys, 2022-02-08:
It was working perfectly and now will not power on.
Don't know where to start. I bought this second hand a week ago.

forestcaver, 2022-02-08:
It’s a non-commercial licence so there shouldnt be second hand modules.
But it’s diy so download the schematics and board files, break out the multimeter and oscilloscope! Start with the power rails and rails regulated from them…. Check over the board under high magnification. If the mcu is alive then run the on chip debugger maybe?
Sometimes the mcu doesnt start if the power on slew is too great - you could try a different psu or add a mic803…. I’ve seen a couple of cases of firmware becoming corrupted but that seems unlikely…
There are loads of possibilities why it may not work and you’ll need to narrow it down….

orys, 2022-02-08:
My friend I bought it from someone. Sorry.
I have opened it up but I'm not an electrician and at 49 I barely see to solder anymore such small components.
I would be grateful if I cound send it to someone to fix it for me and I pay him for his labour.
I got this to start a new system and my dreams are dying right now. :(